MarvellEmbeddedProcessors/mainline-public #62: PM: CPU Idle doesn't work on Armada 38x

gclement, 2015-03-30:
Under some circumstance (PCIe related  but not only), kernel hangs when CPU idle is enabled.

tpetazzoni, 2015-05-23:
Gregory, isn't this bug "fixed" now that we have completely disabled cpuidle on Armada 38x?

gclement, 2015-05-25:
I would like to have a proper solution for this bug. Either completely removing CPU Idle if it does not work at all, or being able to enable for some configuration. The best solution would be to find a fix or a work around.

stsp, 2015-05-26:
Why is this not a problem with armada-XP?
I am pretty sure I have that on armada-XP,
but only in DEEP_IDLE state.

This patch:
http://lkml.iu.edu/hypermail/linux/kernel/1503.1/05016.html
prevents entering DEEP_IDLE, and the problem
no longer appears. Is it the same on Armada 38x?

tpetazzoni, 2015-06-10:
@stsp No, the problem shouldn't appear on Armada XP. Armada XP uses the PJ4B-MP CPU core, while Armada 38x uses Cortex-A9. The specific reason why cpuidle is problematic on Armada 38x is because when HW I/O coherency is enabled, we are not allowed to do PL310 maintenance operations through MMIO registers due to a HW deadlock between PL310 and PCIe. This problem does not affect Armada XP.

tpetazzoni, 2015-06-10:
@gclement In my opinion, it's an issue that will never be fixed. So instead of keeping an issue opened forever, I'd prefer to close it, and reopen it later if Marvell ever finds a fix for it which I believe is quite unlikely.

stsp, 2015-06-10:
Thanks for info.
Then the bug 64 needs a separate explanation.
Hope you can reproduce it.

tpetazzoni, 2015-06-12:
Non-working cpuidle on Armada 38x is currently a hardware issue, for which we're waiting for a proposed workaround. There's no point in keeping a bug tracker entry opened for that.

closed 2015-06-12 by tpetazzoni