MarvellEmbeddedProcessors/mainline-public #64: PCI: On Armada XP, sata transfer over PCIe hang

gclement, 2015-03-30:
basically, any serious PCIe activity seems to be hanging the system. Most of the time it's a hard lockup (no sysrq, the HW debugger says the cores are around 0xffff0000), a few times sysrq was still working, but not showing anything useful.

A simple:
     for i in $(seq 0 64) ; do dd if=/dev/urandom of=file$i bs=1M count=16 ; done
executed in an ext4 filesystem mounted on a single SATA drive connected with a Marvell SATA PCIe card, will hang the system. And  it's reproducible.
It can also reproduced by doing:
     md5sum -c MD5SUMS

stsp, 2015-04-23:
Any updates to this?
Seems like I am having this too with the
vanilla kernel but not with marvell's 3.2 kernel.
I am using mvsas driver (my board has sas controller),
and on hdd activity there is a lock-up.

stsp, 2015-04-23:
Also this may be related:
http://www.spinics.net/lists/arm-kernel/msg411431.html

stsp, 2015-05-14:
By reverting the aforementioned patch and
also disabling pmsu, one of our engineers seem
to have got a stable system, no PCIe hangs.
Can this help?

tpetazzoni, 2015-05-23:
I will be taking care of this issue next week.

stsp, 2015-05-23:
It seems the hang happens only when
the pmsu is in DEEP_IDLE state.
There is a patch:
http://lkml.iu.edu/hypermail/linux/kernel/1503.1/05016.html
With this patch we do not observe ever
entering the DEEP_IDLE state, end neither
the lock-up. Of course I suspect this is not
a real fix at all. Maybe it will enter DEEP_IDLE
state under some other conditions and will
hang again.

tpetazzoni, 2015-05-25:
Reverting http://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/commit/drivers/bus/mvebu-mbus.c?id=1737cac6936938a9bc52c03c4a3ff2032c702fa5 only should fix the problem. I don't really see why reverting another patch would be needed.

Are you working on Armada XP or Armada 38x ?

stsp, 2015-05-25:
I am on armada-XP, but the difference is that
I use mvsas controller instead of sata.
Please note that the patch
http://lkml.iu.edu/hypermail/linux/kernel/1503.1/05016.html
may probably already be applied for you, so you
may not notice the problem because of that.

tpetazzoni, 2015-05-28:
I sent a patch which fixes DMA problems on Armada XP, see http://lists.infradead.org/pipermail/linux-arm-kernel/2015-May/346937.html.

I will test after reverting the patch http://lkml.iu.edu/hypermail/linux/kernel/1503.1/05016.html, to see if I can reproduce the issue reported by @stsp

tpetazzoni, 2015-05-28:
I just did a test after reverting "cpuidle: mvebu: Update cpuidle thresholds for Armada XP SOCs", running bonnie++ with a 8GB size on an ext4 filesystem mounted over a PCIe SATA card, and it worked alright.

I'll close the bug for now because I believe things are working. If not, don't hesitate to reopen, but please provide a detailed test case for us to reproduce the problem.

stsp, 2015-05-28:
Thomas, do you have a git tree on which I can
test things to make sure we are testing the same
thing? For now I am just collecting the patches
from here and there to make things barely working,
so our tests do not match.

stsp, 2015-05-28:
Please note that I can't re-open this bug
(not that I wanted, but still).
You might have locked that possibility somehow.

tpetazzoni, 2015-05-28:
@stsp Can you test with the branch at https://github.com/MISL-EBU-System-SW/mainline-public/tree/4.1/mvebu-mbus-fixes-new, using mvebu_v7_defconfig as the configuration?

stsp, 2015-05-28:
No because your defconfig doesn't support mvsas
(and many other things I need). I'll be testing your
branch with my config now.

stsp, 2015-05-28:
OK, I want to re-open this bug because this command:
dd if=/dev/sda of=/dev/null bs=1M count=1000000 iflag=direct
immediately hangs the system.
Here's the git log so that you know what am I up to:

commit c0091b649bcc4462b94e8913a5c263dc4b3382fa
Author: stsp
Date:   Thu May 28 13:35:29 2015 +0300

    Revert "ARM: Blacklist GCC 4.8.0 to GCC 4.8.2 - PR58854"
    
    This reverts commit 7fc150543c73de71859631c8a6b17e3067fe7617.

commit 8af2007c8ccb95a7a0c04d6b290830a6061fe8f7
Author: stsp
Date:   Thu May 28 13:32:23 2015 +0300

    Revert "cpuidle: mvebu: Update cpuidle thresholds for Armada XP SOCs"
    
    This reverts commit ce6031c89a35cffd5a5992b08377b77f49a004b9.

commit 00f8d638b33d3243c8720675024f7a670bb82efd
Author: Thomas Petazzoni <thomas.petazzoni@free-electrons.com>
Date:   Wed May 27 10:42:00 2015 +0200

    bus: mvebu-mbus: add mv_mbus_dram_info_nooverlap()

stsp, 2015-05-28:
I will now mail you my config and dts as it seems
github doesn't support file attachments.

tpetazzoni, 2015-05-28:
Can you confirm on which kernel version / branch exactly you can reproduce this issue? Can you test with the internal SATA ?

stsp, 2015-05-28:
> Can you confirm on which kernel version / branch exactly you can reproduce this issue?

See the git log above: I am on commit 00f8d63, which
is of the branch 4.1/mvebu-mbus-fixes-new.

> Can you test with the internal SATA ?

I'll try, this will take a few days perhaps.

Can you try my binary kernel I've mailed you or
my config? Maybe you'll reproduce...

stsp, 2015-05-28:
> Can you test with the internal SATA ?

Oh, if you mail me the u-boot image for armada-xp-gp
board, then yes! :) So far I wasn't able to build anything
that works.

stsp, 2015-06-02:
Hello Thomas.
I've finally booted my RD-78460-SERVER board and
reproduced the lock-up there as well, with exactly same
kernel. So mv-sata is affected as well.
What can I do to help you reproduce the problem?

tpetazzoni, 2015-06-02:
I'm currently away this week. But maybe @gclement can try to reproduce. If not, I'll try to reproduce next week.

stsp, 2015-06-02:
Thanks!
If Gregory also needs my config or binary kernel
with appended DT - please tell me and I'll mail it.

In fact, I'd like to test with defconfig and default DT
to reduce the problem of reproducing this.
Could you please hint me what DT should be used
for RD-78460-SERVER board?

stsp, 2015-06-02:
OK, reproduced with defconfig and armada-xp-gp dts.

gclement, 2015-06-02:
OK so I will try on the Armada XP GP board

stsp, 2015-06-02:
The test command is:
dd if=/dev/sda of=/dev/null bs=1M count=1000000 iflag=direct

My board is RD-78460-SERVER, not sure how it differs
from XP GP.

stsp, 2015-06-02:
Please don't forget about
git revert ce6031c89

gclement, 2015-06-03:
I didn't mange to reproduce it, I had I/O error before (But it is possible that my board was damaged).
Could you sent the kernel image and your .config?

stsp, 2015-06-03:
Will do tomorrow, thanks for trying.
Although since I've used the default config
and default DT, it will unlikely help...
Just to make sure, you reverted  ce6031c before
testing, right?
Do you have the RD-78460-SERVER reference
board somewhere? Also can we somehow compare
the CPU cut/stepping/whatever? Maybe your soc
is somehow a bit newer than mine?

stsp, 2015-06-03:
Also please add "pm_disable" to your bootargs.
That's how I tested.

gclement, 2015-06-04:
Having your binary may help to be sure that it was not a compiler related problem.
From U-Boot I got:
Board: RD-AXP-GP rev 1.0
SoC:   MV78460 B0
       running 4 CPUs
       Custom configuration
CPU:   Marvell PJ4B (584) v7 (Rev 2) LE.
I didn't used pm_disable, I will try with this one

stsp, 2015-06-04:
> I didn't used pm_disable, I will try with this one

Please don't - I already tested and made sure it is irrelevant.
I'll mail you everything very soon.

stsp, 2015-06-04:
Gregory, I mailed you my kernel (with DT), config
and u-boot. Please don't tell me its still not reproduceable. :))

Here's my /proc/cpuinfo:

processor       : 0
model name      : ARMv7 Processor rev 2 (v7l)
BogoMIPS        : 1594.16
Features        : half thumb fastmult vfp edsp thumbee vfpv3 tls idiva idivt vfpd32 lpae
CPU implementer : 0x56
CPU architecture: 7
CPU variant     : 0x2
CPU part        : 0x584
CPU revision    : 2

(4 CPUs of that kind)

stsp, 2015-06-04:
Board: RD-AXP-GP rev 1.0                                                        
SoC:   MV78460 B0                                                               
       running 4 CPUs                                                           
       Custom configuration                                                     
CPU:   Marvell PJ4B (584) v7 (Rev 2) LE 

That's what I get from u-boot.
So either our CPUs are similar, or u-boot just configured
that way and doesn't read the actual info.

gclement, 2015-06-04:
Finally I managed to reproduce the issue. Actually, I didn't use the mvebu_v7_defconfig in the first place.
The other change I made was initially I used a initramfs, now I use an nfsroot.
The last change is that I perform the dd on a small SSD disk on sda whereas initially I used a "real" hard drive on sdb.

stsp, 2015-06-04:
Wow!!!
Great news!
Please keep me updated.

stsp, 2015-06-08:
Any news?
I would be especially interested to know with
what exactly config you were not able to reproduce
it initially? Is there some macigal config that just works?

tpetazzoni, 2015-07-01:
I've tested with Linux 4.1, mvebu_v7_defconfig with no change except using an initramfs, on Armada XP GP, with a 8 GB SSD drive, and dd works perfectly fine:

# /usr/bin/dd if=/dev/sda of=/dev/null bs=1M count=1000000 iflag=direct
7627+1 records in
7627+1 records out
7998455808 bytes (8.0 GB) copied, 163.763 s, 48.8 MB/s
# 

I.e, no problem at all.

stsp, 2015-07-01:
Gregory, could you help Thomas reproducing?

Thomas, my testing shows that reproducing with
SSD is more difficult. Do you have HDD?

tpetazzoni, 2015-07-01:
@stsp Test started on a 500 GB HDD.

stsp, 2015-07-01:
If you don't reproduce it in 10 seconds, you probably
can't, so don't spend too much time on testing.
Hope Gregory can help... He said he can reproduce.

stsp, 2015-07-01:
Just to make sure, you reverted ce6031c before
testing, right?

tpetazzoni, 2015-07-01:
No, I did not revert ce6031c before testing. So what you say is that with the current kernel it works thanks to the cpuidle threshold change, but if we revert back to the more aggressive thresholds, it no longer works?

And indeed, I have been running the dd test on a HDD since 20 minutes, and the board is still alive so far.

stsp, 2015-07-01:
Yes, exactly.
I mentioned many times in this thread, that commit
ce6031c obscures the problem by _never_ entering
DEEP_IDLE state. That was the trick to hide the real
bug perhaps (it comes from some marvell patch).
You need to revert that commit, and, as soon as you
in DEEP_IDLE state, you lock up.

stsp, 2015-07-01:
> And indeed, I have been running the dd test on a HDD since 20 minutes, and the board is still alive so far.

Don't spend much time on that.
Better to do 5  1-minute tests changing SSD/HDD
than 1 20-minute test, because the problem either
reproduces immediately, or not at all.
Though I hope after reverting the commit you'll have
no problems reproducing.

tpetazzoni, 2015-07-02:
I confirm I can reproduce the issue after reverting ce6031c.

stsp, 2015-07-02:
Yeah! Great news, thank you.
I wish such a patch is in fact reverted: it secretly
disables DEEP_IDLE, which is not only obscuring
the bug but is also deteriorating the power saving.
Such things must be clearly changelogged, or never
done. Gross.

tpetazzoni, 2015-07-02:
@stsp Contrary to what you say, ce6031c does not make it impossible to enter the idle state. Indeed in the default mvebu_v7_defconfig, NO_HZ is not enabled, so we have a tick every 10 ms, so we never enter DEEP_IDLE. However, if you enable NO_HZ, the 10 ms tick is disabled when the system is idle. And we indeed enter DEEP_IDLE, both when the system is idle *and* when the dd program is running. But I can't reproduce the system freeze in this case.

tpetazzoni, 2015-07-02:
@stsp Correction: the freeze just occured. It just took longer.

stsp, 2015-07-02:
Thanks for an update.
I was about to start testing with NO_HZ.
Now this won't be needed.

stsp, 2015-07-02:
Now this means you can reproduce the
bug even without any reverts - very good!

tpetazzoni, 2015-07-02:
Reverting the patch, I could reproduce the problem faster, and this time got a:

[  218.943155] Unable to handle kernel NULL pointer dereference at virtual address 00000000

but nothing else.

stsp, 2015-07-02:
I think these are meaningless.
I've seen random addresses.
And they seems to be printed out very slowly,
as if the CPU is overloaded by some level-triggered irq.

tpetazzoni, 2015-07-03:
@stsp Can you try with https://github.com/MISL-EBU-System-SW/mainline-public/tree/4.1/cpuidle-test-axp ? This is 4.1 + the cpuidle patch reverted + switch back from automatic I/O sync barriers to explicit ones (note: we know there are issues when using explicit barriers, which is why we switched to automatic).

I've tried here, and it does seem to be better, I haven't reproduce so far. But another independent testing can help. Of course, from this branch, you can get rid of the top two commits and see if you can reproduce the issue.

Thanks for your contribution!

tpetazzoni, 2015-07-03:
@stsp With the above branch, I have been able to read a full 500 GB hard disk with no problem. So it seems like the interaction of DEEP_IDLE with automatic I/O sync barriers is causing problems.

stsp, 2015-07-03:
No help (tested with mvsas).
I also enabled the NO_HZ, and now it locks up
even before the booting process is completed,
no need to run a test-case.
I'll be mailing you my kernel image.

tpetazzoni, 2015-07-03:
@stsp You tested https://github.com/MISL-EBU-System-SW/mainline-public/tree/4.1/cpuidle-test-axp, or https://github.com/MISL-EBU-System-SW/mainline-public/tree/4.1/cpuidle-test-axp with the two last patches reverted ?

tpetazzoni, 2015-07-03:
Also, from a NO_HZ point of view, what I have is:

```
CONFIG_NO_HZ_COMMON=y
# CONFIG_HZ_PERIODIC is not set
CONFIG_NO_HZ_IDLE=y
# CONFIG_NO_HZ_FULL is not set
# CONFIG_NO_HZ is not set
```

I have no idea whether CONFIG_NO_HZ_FULL can work or not.

stsp, 2015-07-03:
I tested both without 2 last patches (to make
sure I am on the right track), and then with all
patches (that are reverts).
Anyway, I'll mail you my kernel so you can make
sure I did the right thing. Config is available in
/proc/config.gz

stsp, 2015-07-03:
I mailed you my kernel.
Here's the git log:

commit c5e1bae15f6d40ef770159d200fd6969c99ae99f
Author: root <root@node-010-117.dcenter.rikor.com>
Date:   Fri Jul 3 16:18:51 2015 +0300

    Revert "ARM: Blacklist GCC 4.8.0 to GCC 4.8.2 - PR58854"
    
    This reverts commit 7fc150543c73de71859631c8a6b17e3067fe7617.

commit ba52c7abe9c0b51d8e403fae65ca3cf0916a3cfb
Author: Thomas Petazzoni <thomas.petazzoni@free-electrons.com>
Date:   Fri Jul 3 10:27:22 2015 +0200

    Revert "bus: mvebu-mbus: use automatic I/O synchronization barriers"
    
    This reverts commit a0b5cd4ac2d6542d524d8063961bf914b5df1efa.

stsp, 2015-07-03:
There is just one additional revert on top of your branch.

tpetazzoni, 2015-07-03:
I'm not able to boot your kernel, it doesn't mount my NFS root filesystem, while it does with my kernel:

```
[  105.308177] VFS: Unable to mount root fs via NFS, trying floppy.
[  105.315304] VFS: Cannot open root device "nfs" or unknown-block(2,0): error -6
[  105.322586] Please append a correct "root=" boot option; here are the available partitions:
[  105.331002] 1f00           16384 mtdblock0  (driver?)
[  105.336083] 1f01           16384 mtdblock1  (driver?)
[  105.341198] 0800         7810992 sda  driver: sd
[  105.345841]   0801          248832 sda1 0005d5f1-01
[  105.350780]   0802         3906560 sda2 0005d5f1-02
[  105.355684]   0803               1 sda3 
[  105.359665]   0804         2968411 sda4 0005d5f1-04
[  105.364568]   0805          683008 sda5 0005d5f1-05
[  105.369508] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(2,0)
[  105.377807] CPU: 1 PID: 1 Comm: swapper/0 Not tainted 4.1.0-78215-gc5e1bae #6
```

The kernel command line is unchanged between booting my kernel and yours.

[37 more replies]