MarvellEmbeddedProcessors/mainline-public #67: Optimize L2 cache performance and power consumption

gclement, 2015-03-30:
- on LSP some specific initalizationis done for the L2CC on Armada 375 and Armada 38x
- check if the property "wt-override" is used on purpose for Armada 370 and Armada XP

tpetazzoni, 2015-03-30:
For the 375/38x part, I've already started working on it.

tpetazzoni, 2015-06-12:
For Armada 375, 38x and 39x, patches have been submitted to the LAKML: http://lists.infradead.org/pipermail/linux-arm-kernel/2015-June/350003.html, http://lists.infradead.org/pipermail/linux-arm-kernel/2015-June/350000.html.

tpetazzoni, 2016-02-10:
The fine tuning of the L2 on 375, 38x and 39x has been merged as part of c8f5a878e55459b78bac273f6492bbc91475beb5.

However the L1 prefetch enabling has not been merged so far. I posted it again on January 2016 (http://lists.infradead.org/pipermail/linux-arm-kernel/2016-January/402819.html) but there was no feedback.

Gregory, do you remember what was the deal with the wt-override property on 370/XP ?

gclement, 2016-02-10:
I don't find any answer about the use of wt-override property on 370/XP.