MarvellEmbeddedProcessors/mainline-public #71: kernel hang on any Oops

stsp, 2015-04-24:
Hello.

Not sure if this is the right place to report the
armada-xp related bugs, but I hope so.

It seems any Oops results in a kernel panic and
machine freeze. To reproduce, you only do:
echo c >/proc/sysrq-trigger
Normally this should Oops and kill the bash process.
On armada-xp it causes a kernel panic instead, and
the freeze.

gclement, 2015-04-28:
the linux-arm-kernel mailing list should be an ever better place to report this. I have just try it on an Armada XP based board and I managed to reproduce it. But on another very different platform (an OMAP based one) I observed the same behavior: kernel panic. So it might be not related to Armada XP but maybe to ARM. That's why it would worth to report this on the mailing list.

stsp, 2015-04-28:
Even worse: I've just locked up my PC with this!
So the regression looks generic, not even arm-specific.
I reported it here:
https://bugzilla.kernel.org/show_bug.cgi?id=97461
Sorry for bothering you with that. I have not expected
such a serious regressions can survive in generic
kernel code for so long...

closed 2015-04-28 by stsp