MarvellEmbeddedProcessors/mainline-public #73: spi broken on armada-XP

stsp, 2015-05-27:
I have armada-XP board with an spi flash that
is identified this way:
m25p80 spi0.0: n25q128a13 (16384 Kbytes)
When I read it, the first byte in the transfer
appear to be random, but all subsequent bytes
are fine.
This problem does not appear on marvell kernel.

gclement, 2015-05-27:
Could you test the last series sent about the SPI ?
http://thread.gmane.org/gmane.linux.ports.arm.kernel/416054

gclement, 2015-05-27:
On the Armada XP GP board I manged to reproduce the issue (it use the same SPI flash). With the series applied, the values are no more random and are the same ones that the ones read under U-boot.

So this series fix the issues, actually the patch "spi: orion: Fix maximum baud rates for Armada 370/XP" is enough. It will be part of the 4.2 release and should be backported on all the stable kernel.

closed 2015-05-27 by gclement

stsp, 2015-05-27:
Thanks Gregory!
Will try tomorrow.
In a mean time I've found this thread:

>> -- compatible : should be "marvell,orion-spi" or "marvell,armada-370-spi".
>> +- compatible : should be on of the following:
>> +    - "marvell,orion-spi" for the Orion, mv78x00, Kirkwood and Dove SoCs
>> +    - "marvell,armada-370-spi", for the Armada 370 SoCs
>> +    - "marvell,armada-375-spi", for the Armada 375 SoCs
>> +    - "marvell,armada-380-spi", for the Armada 38x SoCs
>> +    - "marvell,armada-390-spi", for the Armada 38x SoCs
> 
> Hi Gregory
> 
> Is that a typ0?
yes it is! :/

IMHO the typo was in that sentence:
>> +- compatible : should be on of the following:

"on" should read as "one"

stsp, 2015-05-28:
Patch tested and works, thanks!

stsp, 2015-10-20:
Gregory, we do still observe the SPI
data corruption, its just much less frequent,
but happens. Were there any further fixes
after this one?

stsp, 2015-10-20:
In fact, I only applied a single patch, based on
this comment of yours:
"Fix maximum baud rates for Armada 370/XP" is enough."
Should I try also the rest?