MarvellEmbeddedProcessors/mainline-public #74: how to enable ECC?

stsp, 2015-06-08:
Hi guys.

Sorry for a support question, I am unable to
enable ECC machinery for armada-xp.
I enabled EDAC in the kernel, but I can't find the
armada-specific driver. edac-util says:
edac-util: Error: No memory controller data found.
All I need is to get the count of ECC errors.
How to do that?

yotamadmon, 2015-06-08:
DRAM ECC or L2 ECC?
DRAM ecc should be enable by U-boot.
ECC counters are 0x1458 (single bit) and 0x145c (double bit).
Thanks,
Yotam

________________________________
Yotam Admon
Staff Application Manager
Controllers and SOC Product Line
EBU Customer Solution Group
Marvell Israel LTD
6 Hamada Street, P.O.B. 192
Yokneam 2069203, ISRAEL
Email: yotam@marvell.com<mailto:yotam@marvell.com>
Office:  +972.3.9703713 - OnNet: 705.3713
Fax:      +972.3.5279786
Mobile: +972.54.2490423
Web site: http://www.marvell.com<http://www.marvell.com/>

This message may contain confidential, proprietary or legally privileged information. The information is intended only for the use of the individual or entity named above. If the reader of this message is not the intended recipient, you are hereby notified that any dissemination, distribution or copying of this communication is strictly prohibited. If you have received this communication in error, please notify us immediately by telephone or by e-mail and delete the message from your computer.

stsp, 2015-06-08:
I meant DRAM ECC.
Is there a driver for armada-xp for edac-utils to work?
Or should I go for a custom solution to read these regs?

tpetazzoni, 2015-06-08:
@stsp Please always indicate which kernel version you are talking about. Since you're posting in the "mainline-public" repository, I assume you're talking about the mainline kernel. And if that's the case, then the answer is easy: there is no support for Armada XP ECC in the mainline kernel, be it for the DRAM or L2 ECC.

This has so far not been a priority, so it has never been on our TODO-list. Contributions welcome!

closed 2015-06-08 by tpetazzoni