MarvellEmbeddedProcessors/mainline-public #75: Fix i2c offloading

tpetazzoni, 2015-07-04:
As Antony Pavlov noticed, there is an unconditional "return false" at the beginning of the mv64xxx_i2c_can_offload() function which makes the offload mechanism never used. We need to remove it, and retest if the I2C works correctly.

eranba, 2015-07-16:
Just removing the "return false" will probably won't be enough  to enable the offloading feature for certain devices(A38x, A39x etc), right ?
Seems like the "offload_enabled" flag should be set as well.

tpetazzoni, 2016-02-10:
The offload_enabled flag is already set when the compatible string is marvell,mv78230-i2c.

The unconditional "return false" was removed as part of commit 0729a04977d497cf66234fd7f900ddcec3ef1c52, so this issue can be closed.

closed 2016-02-10 by tpetazzoni