MarvellEmbeddedProcessors/mainline-public #76: fw_setenv works unreliably

stsp, 2015-10-28:
I have spi flash n25q128a13.
When using fw_setenv, most of the times the
environment gets corrupted. The important thing
is to always write different strings, like
fw_setenv tst test1
fw_setenv tst test2
fw_setenv tst test3
If you write always the same string, the problem
does not happen, which makes me to suspect
that erase doesn't work.
The problem does not appear on marvel kernel.

stsp, 2015-10-28:
I fixed the problem, but I don't know how to attach
the patch files to the ticket.

stsp, 2015-10-28:
From 632e0e21d7520929c42eaa4835cc000847c6a736 Mon Sep 17 00:00:00 2001
From: Stas Sergeev <stsp@users.sourceforge.net>
Date: Wed, 28 Oct 2015 19:36:59 +0300
Subject: [PATCH 1/2] spi_nor: use spi_nor_wait_till_ready() before operations

---
 drivers/mtd/spi-nor/spi-nor.c | 16 ++++++++++++++++
 1 file changed, 16 insertions(+)

diff --git a/drivers/mtd/spi-nor/spi-nor.c b/drivers/mtd/spi-nor/spi-nor.c
index b6a5a0c..fd4f6ea 100644
--- a/drivers/mtd/spi-nor/spi-nor.c
+++ b/drivers/mtd/spi-nor/spi-nor.c
@@ -313,6 +313,10 @@ static int spi_nor_erase(struct mtd_info *mtd, struct erase_info *instr)
 	addr = instr->addr;
 	len = instr->len;
 
+	ret = spi_nor_wait_till_ready(nor);
+	if (ret)
+		return ret;
+
 	ret = spi_nor_lock_and_prep(nor, SPI_NOR_OPS_ERASE);
 	if (ret)
 		return ret;
@@ -704,6 +708,10 @@ static int spi_nor_read(struct mtd_info *mtd, loff_t from, size_t len,
 
 	dev_dbg(nor->dev, "from 0x%08x, len %zd\n", (u32)from, len);
 
+	ret = spi_nor_wait_till_ready(nor);
+	if (ret)
+		return ret;
+
 	ret = spi_nor_lock_and_prep(nor, SPI_NOR_OPS_READ);
 	if (ret)
 		return ret;
@@ -723,6 +731,10 @@ static int sst_write(struct mtd_info *mtd, loff_t to, size_t len,
 
 	dev_dbg(nor->dev, "to 0x%08x, len %zd\n", (u32)to, len);
 
+	ret = spi_nor_wait_till_ready(nor);
+	if (ret)
+		return ret;
+
 	ret = spi_nor_lock_and_prep(nor, SPI_NOR_OPS_WRITE);
 	if (ret)
 		return ret;
@@ -794,6 +806,10 @@ static int spi_nor_write(struct mtd_info *mtd, loff_t to, size_t len,
 
 	dev_dbg(nor->dev, "to 0x%08x, len %zd\n", (u32)to, len);
 
+	ret = spi_nor_wait_till_ready(nor);
+	if (ret)
+		return ret;
+
 	ret = spi_nor_lock_and_prep(nor, SPI_NOR_OPS_WRITE);
 	if (ret)
 		return ret;
-- 
1.9.1

stsp, 2015-10-28:
Grr

gclement, 2015-11-06:
I think the best option is to post your patch on the mtd mailing list.

stsp, 2015-11-06:
I posted the patch here:
https://bugzilla.kernel.org/show_bug.cgi?id=106881
mtd list is unhelpful afaik, I tried to post some patches
there in the past.
Since this is likely not related to marvel, lets close this.

ezequielgarcia, 2015-11-06:
Stas,

The MTD mailing list is quite active and responsive as well. I think it would be very interesting to have your fixes :-)

Regarding the two patches you've attached:

1) "m25p80: dont swallow erase return status" is obviously correct

2) "spi_nor: use spi_nor_wait_till_ready() before operations" seems to be papering over some other bug. Putting a wait_till_ready before calling prepare() everywhere doesn't sound like a proper solution (despite it works for you).

Have you noticed that m25p80 doesn't set the prepare hook? It seems to be a no-op
in your case. I bet that not all of your wait_till_ready additions are needed, so maybe you can just leave the just those (or the single one) that fix your issue?

We can help you with the fix upstreaming, in case you are still interested.

stsp, 2015-11-06:
I tried to post a patch there that adds a very simple
tool for marking bad blocks:
http://lists.infradead.org/pipermail/linux-mtd/2014-April/053492.html
and there was no resp.
When I got the more complex patches, I tried to upstream
them via ST:
https://bugzilla.stlinux.com/show_bug.cgi?id=47769#c15
but that didn't work too.
So as of yet I have no place where the mtd patches can
be reviewed or upstreamed. If you can help with that, that
would be great.

As for wait_till_ready() additions: in the older kernels, like
3.2, in m25p80.c all wait_till_ready() precedes the flash
operation. That works fine. But in newer kernels they are
put _after_ the operations, instead of before. I don't know
why this was done, but this doesn't work. My patch just
adds them before the ops, without removing the ones after
the ops (since I don't know why do they exist, but there was
probably a reason). So my patch is likely superfluous, but
should be safe, as it matches the old driver.

To answer your question if only a single addition is enough
for me: yes; before erase. Then it seems to work. But I don't
know why only one addition should be done if the old driver
had all of them. I can do further tests to locate the problem
more precisely, if you suggest what should be tested.

stsp, 2015-11-06:
You can boot with boot.blk, and at one point
reset int1e back to dosemu bios to see if it will
then still boot.

stsp, 2015-11-06:
Oops, wrong window, sorry

ezequielgarcia, 2015-11-07:
> To answer your question if only a single addition is enough
> for me: yes; before erase. Then it seems to work. But I don't
> know why only one addition should be done if the old driver
> had all of them. I can do further tests to locate the problem
> more precisely, if you suggest what should be tested.

Do you have some isolated test that I can run here that exhibits
the bug? (without using fw_setenv). That'll help fix the bug.

stsp, 2015-11-07:
> he bug? (without using fw_setenv).

Very strange requirement. Why without fw_setenv?
What can that be at all? With flash_erase I didn't
hit the problem, and I don't know any other tool.

stsp, 2015-11-17:
@ezequielgarcia I uploaded a much smaller patch.
Could you please review?

ezequielgarcia, 2015-11-17:
@stsp Hm, can't find any uploads.

stsp, 2015-11-17:
https://bugzilla.kernel.org/attachment.cgi?id=193801

ezequielgarcia, 2015-11-17:
Ah, I see. This looks much better. Is this enough to fix all your issues
with the SPI?

I've asked the MTD maintainer (US timezone, so we'll have to wait).
Meanwhile, have you checked your SPI flash datasheet? Maybe it specifies
this is needed?

ezequielgarcia, 2015-11-17:
@stsp Just to check: which repo / branch is your patch applying to?

stsp, 2015-11-17:
Oh, this is some old 4.0.0-rc2 with many patches from
this bugzilla. I'll be having problems to check if the problem
is still present in the latest kernel. :(

> Ah, I see. This looks much better. Is this enough to fix all your issues
> with the SPI?

Together with SPI patches from @gclement - yes.

I am now looking in datasheet to find something interesting.

ezequielgarcia, 2015-11-17:
@stsp From the N25Q spec:

""
WRITE STATUS REGISTER Command
[..]
When the operation is in progress, the write in progress bit is set to 1. The write enable
latch bit is cleared to 0, whether the operation is successful or not. The status register
and flag status register can be polled for the operation status. When the operation completes,
the write in progress bit is cleared to 0, whether the operation is successful or
not.
""

So effectively the write_sr seems that should be followed by a read_sr.
Your patch seems correct to me.

Want me to prepare an upstream patch? Or do you want to do that yourself?

stsp, 2015-11-17:
Of course I'd like you to!
It is nearly impossible for me because I can
work only on some 4.0-rc2 or have the major
headache upgrading and moving forward all
the patches I have.

ezequielgarcia, 2015-11-17:
@stsp No problem :)

closed 2015-11-17 by ezequielgarcia

stsp, 2015-11-17:
Please in either case keep me in CC.

ezequielgarcia, 2015-11-17:
@stsp: sure

stsp, 2015-12-01:
@ezequielgarcia I was CCed with your initial msg.
No reply as of yet?

stsp, 2015-12-02:
@ezequielgarcia It also seems as if you only posted
one patch from 2?

ezequielgarcia, 2015-12-02:
@stsp Brian will merge the patch sooner or later, don't worry.

Regarding the other patch, m25p80 no longer implements its own erase, so the patch is not needed. Fixed here:

http://git.infradead.org/l2-mtd.git/commit/cd78ea02dc82eb26be8c2fc1cf05adbf94ea8727