MarvellEmbeddedProcessors/mainline-public #81: UART1 on 3720

qume, 2019-05-29:
I have the kernel reporting UART1 / /dev/ttyMV1, the serial port on pins 24 and 26 on the Espressobin board. However I am unable to send or read any data from it.

It is enabled in the device tree.

The TX pin sits at a constant 1.8v, regardless of sending characters to it.

I see from a discussion @Kodest had this working on the ESPRESSObin v5.0.1. I am using the v7 board. Discussion here:
https://github.com/MarvellEmbeddedProcessors/mainline-public/pull/79

One thing to note is setserial reports no IRQ:
/dev/ttyMV1, UART: undefined, Port: 0x0000, IRQ: 0

Here is the dmesg line:
[    2.122362] d0012200.serial: ttyMV1 at MMIO 0xd0012200 (irq = 0, base_baud = 1562500) is a mvebu-uart

miquelraynal, 2019-05-29:
Can you share the diff adding the DT node? Did you point somehow to <&uart2_pins> in a pinctrl-default property? Have you disabled any other pin function that could be enabled on the same MPP pin? (something like I2C/SPI/MMC).

Also, what is your kernel version? Does the driver support the extended UART compatible (marvell,armada-3700-uart-ext)? See for instance ac9a44b80c2b ("serial: mvebu-uart: support extended port registers layout").

qume, 2019-05-31:
I'm using the branch "linux-4.14.76-armada-18.12". Diff at bottom of this note.

I haven't enabled anything else on the pin, however I'm not familiar enough with the whole DT system to be able to check for sure nothing was already enabled. I did confirm that no other pinctrl items contained the same IRQ. And nothing else references uart2_pins. 

>  Did you point somehow to <&uart2_pins> in a pinctrl-default property?
I assume that is what is happening in the diff below?

I checked the driver and it has the patch you referenced applied

The only thing I can think of which isn't just default in the repo, apart from this patch, is configuring /sys/class/gpio support in the kernel build.

```
diff --git a/arch/arm64/boot/dts/marvell/armada-3720-espressobin.dts b/arch/arm64/boot/dts/marvell/armada-3720-espressobin.dts
index 488ff5a19266..b007588187d1 100644
--- a/arch/arm64/boot/dts/marvell/armada-3720-espressobin.dts
+++ b/arch/arm64/boot/dts/marvell/armada-3720-espressobin.dts
@@ -114,6 +114,12 @@
  * - MMC
  */
 
+/* P9 pin 24, 26 */
+&uart1 {
+	pinctrl-names = "default";
+	pinctrl-0 = <&uart2_pins>;
+	status = "okay";
+};
 /* J7 */
 &usb3 {
 	status = "okay";

```

qume, 2019-05-31:
Note I have tested three boards with the same behaviour.

miquelraynal, 2019-06-17:
I just built and run a Marvell 4.14.76 kernel, the interrupts are fine and both ttyMV{0,1} are up.
```
# cat /proc/interrupts 
           CPU0       CPU1       
[...]
  9:        198          0     GICv3  44 Level     d0012000.serial
 10:         60          0     GICv3  45 Level     d0012000.serial
```

The only change in the DT is:

```
diff --git a/arch/arm64/boot/dts/marvell/armada-3720-espressobin.dts b/arch/arm64/boot/dts/marvell/armada-3720-espressobin.dts
index 488ff5a19266..1c21075ec1fc 100644
--- a/arch/arm64/boot/dts/marvell/armada-3720-espressobin.dts
+++ b/arch/arm64/boot/dts/marvell/armada-3720-espressobin.dts
@@ -104,6 +104,12 @@
        status = "okay";
 };
 
+&uart1 {
+       pinctrl-names = "default";
+       pinctrl-0 = <&uart2_pins>;
+       status = "okay";
+};
+
 /*
  * Connector J17 and J18 expose a number of different features. Some pins are
  * multiplexed. This is the case for instance for the following features:
```

Here is [the config file I used](http://code.bulix.org/088rj7-774069).